mzy2240/ESA #1: Update to use PowerWorld's "Concise Variable Names" instead of "Variable Name"

blthayer, 2019-11-19:
It turns out that despite being listed first in the PowerWorld export, the "Variable Name" column contains old/deprecated variable names, and the "Concise Variable Name" column represents the new/preferred variable names.

This definitely isn't critical, but it's probably better to fix this earlier rather than later. This will impact tests, examples, etc.

We should also consider if we want to support both variable names.

Finally, this will require documentation updates (which should be made no matter what to make it clear to our users what we support).

mzy2240, 2019-11-20:
This is a very good find. What do you mean that the "variable name" column contains old/deprecated names? How did you find that? And is there a way to generate a list of valid "concise variable names" or any resource that points to the preferred variable names?

blthayer, 2019-11-20:
When you export the variable names like so:

[image]

you'll get an Excel sheet that looks like:
[image]

It turns out that the "Concise Variable Name" is preferred, while "Variable Name" is legacy. I learned this in the PowerWorld course. So, we should switch to those, and ensure it's well documented/clear to our users that that's what we support.

mzy2240, 2022-02-12:
Currently when `pw_order = True`, ESA will use exactly the same variable names as shown in the model explorer. The tradeoff is, you will need to do type conversion by urself (which is very trivial). People who prefer using the same variable names could use this feature.

mzy2240, 2022-02-12:
Close the issue for now.

closed 2022-02-12 by mzy2240