mzy2240/ESA #21: Implement and test SendToExcel method

blthayer, 2020-03-25:


blthayer, 2020-03-27:
@SmartJingJing123 @mzy2240 
I'm re-opening this for two reasons:
1. Most importantly - we need a unit test for this method. The test should call the method, confirm that the file gets created, and then delete the file (using the tearDown from unittest.TestCase). I can write this test if desired.
2. Less important, but still important :) - the indentation for the method's docstring is not correct. This will cause the API documentation to not come out right. Please see other methods for reference. 

Let me know if you have questions or need help with anything!

mzy2240, 2020-03-27:
@blthayer You are right. I already notified her about that when I saw the commit.

SmartJingJing123, 2020-03-27:
Thanks for the reply. I have fixed the docstring issue, and am working on
the tests.

Best,
Yijing

mzy2240, 2020-03-28:
#d6ad27d

mzy2240, 2020-03-28:
@SmartJingJing123 I just verified and it failed to pass the test.

blthayer, 2020-03-30:
@mzy2240 @SmartJingJing123 We need more tests for this method. Currently the only test that's in place shows that an error is correctly thrown if the method is asked for a bad ObjectType. We also need to ensure it behaves correctly when given valid input. However, I can see how this might be tricky: it looks like the method opens up Excel rather than just writing to an Excel file?

mzy2240, 2020-03-30:
@blthayer You are right. I talked with @SmartJingJing123 about that, and she mentioned she cannot access the opened excel sheet (it is not saved). I recommended her to use the Excel COM object, but she still had a problem accessing it.

blthayer, 2020-03-30:
@mzy2240 @SmartJingJing123 - as discussed in today's meeting, @SmartJingJing123 will document what she attempted in the test method, and update the docstring in the main method to warn users that we were not able to sufficiently test this method. Also, recommend users simply get objects and use pandas to save .csv files.

closed 2020-04-01 by mzy2240