mzy2240/ESA #30: Update top-level README

blthayer, 2020-03-30:
- Make section for documentation
- Add one-liner for pip installation

blthayer, 2020-04-01:
@mzy2240 @SmartJingJing123 - take a look at the top-level README and let me know if you think it's adequate. Again, the reason to keep it slim is to avoid repeating information, which leads to unmaintainable documentation. GitHub does not allow us to use the `include` directive when rendering files on GitHub (such as the README).

closed 2020-04-01 by blthayer

mzy2240, 2020-04-01:
Since you mentioned that for full details check the documents, I would recommend, in the installation section, just put the `pip install esa`.

blthayer, 2020-04-01:
@mzy2240 - sure thing, I can do that if you think it's better. I put the "only binary" stuff in there since I think that'll minimize the errors users receive. However, maybe that's punishment for not checking out the documentation :)