mzy2240/ESA #31: Draft statement of need

blthayer, 2020-03-30:
Discuss why one would use an automation tool like ESA instead of directly using PowerWorld on its own.

mzy2240, 2020-04-07:
- Getting data from a power system model for other application, like graph analysis or signal processing
- Automate some parameter search process, like dynamic model tunning
- Provide a bridge between the power system simulation and the machine learning application

You guys are welcomed to provide more reasons @SmartJingJing123 @blthayer

blthayer, 2020-04-10:
I like these. I think some other ideas might be:
- Co-simulation with other domains such as transportation, communication, or natural gas
- General task automation

blthayer, 2020-04-10:
Update the why use ESA documentation to include this stuff

mzy2240, 2020-04-15:
Updated in #7ae001e.

closed 2020-04-15 by mzy2240