mzy2240/ESA #33: Write draft of JOSS paper

blthayer, 2020-03-30:


blthayer, 2020-04-08:
@mzy2240 @SmartJingJing123 

I've written a rough draft. It'll likely need a good bit of revising, and we need to add more projects that use ESA. 

I'm going to reassign this task to you two - please read it through and revise. Feel free to either directly make revisions, or you can make notes and we can talk about revisions later.

You might find it nicer to read a .pdf of the paper. Simply put the GitHub repo URL and `develop` branch into the entry boxes here:
https://whedon.theoj.org/

I've already ensured it compiles correctly.

blthayer, 2020-04-10:
@mzy2240 - please add equal contribution note to acknowledgements section.

blthayer, 2020-04-10:
@mzy2240 - I made some modifications to your addition. Can you please take a look?

mzy2240, 2020-04-10:
It looks great. Thank you!

mzy2240, 2020-04-14:
@blthayer I think now the paper looks good now. However, one problem is that inside the bib section some repeated names are replaced by dashed lines. Though it is the default format of IEEE and it is accepted (or recommended) by JOSS (check here https://github.com/openjournals/joss-reviews/issues/384#issuecomment-332239884), I still prefer to always have the full name instead of lines. Do you mind to try to fix that? I have tried several ways but none of them work.

blthayer, 2020-04-14:
@mzy2240 - I agree and don't necessarily like the dashed lines. Though if it's the default style maybe we should just go with it?

blthayer, 2020-04-14:
@mzy2240 - can you post the command you used to compile the paper? I'm guessing you used pandoc? The current .pdf in the repo seems to be missing the logo - is that expected?

blthayer, 2020-04-14:
@mzy2240 - just an FYI - when I compile with the JOSS preview whedon tool thingy, none of your tweaks (like the footnote) come through. So I doubt that particular service is actually using your latex.template, logo.png, etc. files.

mzy2240, 2020-04-14:
> @mzy2240 - can you post the command you used to compile the paper? I'm guessing you used pandoc? The current .pdf in the repo seems to be missing the logo - is that expected?

Check the makefile file. The command I am using is inside the file.

mzy2240, 2020-04-14:
> @mzy2240 - just an FYI - when I compile with the JOSS preview whedon tool thingy, none of your tweaks (like the footnote) come through. So I doubt that particular service is actually using your latex.template, logo.png, etc. files.

The website does not utilize the template, but I am sure they will use when they compile the actual paper, cause some of the published one also use the template, like this one https://github.com/JuliaEarth/GeoStats.jl/tree/master/paper. The corresponding pdf is here: https://joss.theoj.org/papers/10.21105/joss.00692, you can notice that the author name has a footnote, which is not possible to generate just by the markdown file.

blthayer, 2020-04-14:
@mzy2240 - thank you! As far as I can tell, JOSS's pandoc command can be found here:

https://github.com/openjournals/whedon/blob/6dc0baef9c6c899932ffa179d2d491a6b3cc5752/lib/whedon/compilers.rb#L160

And the latex template they use can be found here:
https://github.com/openjournals/whedon/blob/6dc0baef9c6c899932ffa179d2d491a6b3cc5752/resources/latex.template

I'll update the Makefile to include as many of the options used by JOSS as possible. @mzy2240 - can you ensure that our latex.template file is as up to date as possible? Note sure where you obtained the one we have, but we should probably start with the latest from JOSS and then tweak to include our notes.

While you're at it - could you please make a README.rst file in the `joss` directory noting where you got each file and how it's used? For example, I have no clue how to use that Makefile without installing a ton of software onto my PC (like Visual Studio). Maybe we should just have a `make.bat` file that runs the pandoc command? It seems like that's basically all that the Makefile does...

blthayer, 2020-04-14:
@mzy2240 - update: I think this link more accurately describes the pandoc command:

https://github.com/openjournals/whedon/blob/6dc0baef9c6c899932ffa179d2d491a6b3cc5752/lib/whedon/compilers.rb#L340

mzy2240, 2020-04-14:
> @mzy2240 - thank you! As far as I can tell, JOSS's pandoc command can be found here: [...]

I get the latex.template from here: https://github.com/openjournals/whedon/tree/master/resources/joss and it is mentioned in some issues in the JOSS repository.

blthayer, 2020-04-15:
Closing this issue since the draft is complete.

closed 2020-04-15 by blthayer