mzy2240/ESA #34: Review and revise updated documentation

blthayer, 2020-03-30:
1. Git pull
2. Build documentation
3. Use web browser to view local documentation
4. Read through everything, fix any issues
5. Propose revisions that help us meet the JOSS criteria

Please make a branch for any changes, and use a pull request to get changes reviewed. Assign @blthayer to the pull request.

blthayer, 2020-04-01:
@mzy2240 @SmartJingJing123 - I recommend pulling the latest from develop and building the documentation locally for this task so that you have the most up-to-date docs.

blthayer, 2020-04-13:
@SmartJingJing123 - it looks like you may have accidentally committed to the `master` branch instead of the `develop` branch. I'll go ahead and try my hand using `git cherry-pick` to move those three commits you made over to the `develop` branch.

blthayer, 2020-04-13:
@SmartJingJing123 - I was able to easily get your three commits "cherry-picked" over to `develop`. Please do future work in the `develop` branch. Thank you!

SmartJingJing123, 2020-04-13:
@blthayer When I was trying to commit, I saw the other snippets are on the `master` branch. But maybe we need to commit to `develop` branch at first. Sorry for the misunderstanding and thanks for your help!

blthayer, 2020-04-13:
@SmartJingJing123 - no worries. In general, we do work in the `develop` branch, and then when we've done enough work for a public release we then merge the work from `develop` into `master` and prepare a PyPi release (and increment the version number, rebuild the documentation, etc.).

I was looking at your snippet - I think it'll be a great addition to the documentation! There are a few clean-up items that are needed:

1. If you could please ensure that all text stays within the 72 character column? This matches PEP-8, but more importantly, makes it so that we don't have to scroll sideways when we have multiple documents up side-by-side
2. Ensure white space is correct. ReStructredText is like Python in that whitespace is really important. Right now, on line 27 it says `Solve the power flow:`, however, it's lacking a space/newline character from the code block above it.
3. Please run `tests/test_snippets.py` and ensure that the snippet passes testing.
4. It might be useful to save your plot to a `.png` file and then include it in your example. I would save the `.png` file right in the `snippets` directory, and add a note about the file to `docs/rst/snippets/README.rst` in the appropriate location. Then, in your file (`line_loading_matplotlib_2000.rst`) you can simply use the rst `include` directive to include the image. I haven't included an image myself yet, so you might need to do some googling to figure out how to make it work. Note that PyCharm has a preview functionality which will automatically render the .rst document. There are some buttons toward the top-right of PyCharm that can be used to toggle your view between "Editor only," "Editor and Preview," and "Preview only."

blthayer, 2020-04-13:
@SmartJingJing123 - I thought of one more thing that needs done for this task: please add `matplotlib` to the testing requirements in the top-level `setup.py` file. You'll notice `networkx` is also listed there. This way, when others want to run all the ESA tests they have all the prerequisites installed from the get-go.

SmartJingJing123, 2020-04-13:
@blthayer You're correct! Have done that and hopefully, this is the final commit for this task.

blthayer, 2020-04-14:
@SmartJingJing123 - I've made some tweaks to your example in the following commits:
- e4ffcf165e8be25dfa4fb7a60375dbf0c8f7ca88
- 58aa57f323fd83f9ca60ad3dda619f8a32e91343
- f5f1ee9ed6019a31c73ed9dc5e58e1e535ec01bf

I wrote an excessively detailed commit message on the first one for you. The other two commits are quick and simple. If you could please review them, that would be great.

Please let me know if you have any questions or want to discuss anything.

SmartJingJing123, 2020-04-14:
@blthayer - Thanks for the commits. I think they are pretty good!

closed 2020-04-14 by blthayer