mzy2240/ESA #35: Contact PowerWorld, request ESA be listed on their website

blthayer, 2020-03-30:
To maximize our visibility, we should see if we can get PowerWorld to link to ESA in their documentation and on their website. We can do this as one of the last tasks related to the JOSS submission.

blthayer, 2020-04-01:
@mzy2240 - can you either email me Caroline's email address or email her related to this? We might want to wait until after we've merged develop into master, bumped the version to 1.0, and added the latest on PyPi.

mzy2240, 2020-04-01:
I will email her shortly. I will also share her email to you via email.

mzy2240, 2020-04-01:
Just emailed. Will close the issue once a response is received.

blthayer, 2020-04-04:
PowerWorld, for whatever reason, doesn't want to list our tool. Closing.

closed 2020-04-04 by blthayer