mzy2240/ESA #36: Run all tests with different versions of Python on different machines

blthayer, 2020-04-02:
Run tests with Python versions 3.5, 3.6, 3.7, and 3.8. @blthayer, @mzy2240, and @SmartJingJing123 should all run the tests with all the Python versions to ensure everything is working as expected.

This should be one of the last things we do before submitting the JOSS paper.

blthayer, 2020-04-02:
@mzy2240 @SmartJingJing123 - check out the new script I added to make this all happen automatically.

mzy2240, 2020-04-02:
Passed all the tests with versions from 3.5 to 3.8. Numpy version also matters, but not sure about the minimum acceptable version. If the user is using the 3.5 with Anaconda, he/she might have some problems using the module cause two tests failed due to the unexpected output of pandas/numpy. To make it work with 3.5, the user should use pip or manually install the dependencies (pywin32, pypiwin32, numpy, pandas).

blthayer, 2020-04-02:
@mzy2240 - do you think we should attempt to find the minimum acceptable version of numpy? If so, would you be willing to do this? Alternatively, you can pick a random version that's a few versions old (maybe a year or so old?), and if it works, we can just set that as the minimum. Not perfect, but saves us from endless trial and error with Numpy version.

blthayer, 2020-04-02:
@mzy2240 - did my script work okay for you?

blthayer, 2020-04-06:
@SmartJingJing123 @mzy2240 - when you've both been able to run the ``run_tests_for_all_python_versions.py`` script successfully, please let me know and then we can close this issue.

SmartJingJing123, 2020-04-06:
@blthayer Have run the script and get the results. The results are all "OK" in "test_results_36/37/38", but Python 3.5 failed the test. I have attached all the results got from tests FYI.
[Results from test all python versions.zip](https://github.com/mzy2240/ESA/files/4440887/Results.from.test.all.python.versions.zip)

mzy2240, 2020-04-07:
@blthayer Failed to pass the 3.6 and 3.8. I guess these are due to the fact that the older version of ESA has been installed in both envs (or older version of dependencies have been installed). Have you considered forcely install the new version of ESA or dependencies?

blthayer, 2020-04-07:
@SmartJingJing123 - for Python 3.5, did you see any error messages in the console?

@mzy2240 and @SmartJingJing123 if you run the script like so:

```python run_tests_for_all_python_versions.py -h```

You'll get output like so:

```
usage: run_tests_for_all_python_versions.py [-h] [--local] [--fresh] python_install_dir

positional arguments:
  python_install_dir  Full path to directory where all Python installations exist.

optional arguments:
  -h, --help          show this help message and exit
  --local             Set this flag to install ESA locally from source. If this flag is not set, ESA will be installed
                      from PyPi.
  --fresh             Set this flag to force fresh virtual environments and packages. Existing virtual environments
                      from a previous run of this script will be removed, and packages will be installed with the
                      --force-upgrade flag.
```

If you could please run with the ``--fresh`` flag, I would appreciate it. So, the command will look like:

```
python run_tests_for_all_python_versions.py <python_install_dir> --fresh
```

blthayer, 2020-04-07:
@mzy2240 - it's also worth noting that the tests are run from the tests that are present locally. So, you should stash any local changes (``git stash``) then checkout the master branch, run the script, and then "unstash" your changes afterwards (``git stash pop``).

SmartJingJing123, 2020-04-07:
@blthayer  I re-run the tests with "--fresh"and get a similar result with yesterday. For Python version 3.5, the error messages can be divided into two types:
1. The first type is related to python packages that are not downloaded correctly. Examples are: 
ImportError: No module named 'numpy' & ImportError: No module named 'pandas'
2. The second type is related to esa. Examples are:
ameError: name 'saw' is not defined & NameError: name 'line_key_fields' is not defined.

blthayer, 2020-04-07:
@SmartJingJing123 - thank you! Can you provide the output you got in the terminal? It seems like things aren't installing correctly for you (which then causes the tests to fail).

blthayer, 2020-04-07:
@SmartJingJing123 - I think this may have to do with the Chinese characters in your path. Can you try again from a different location that only has ASCII characters? 

The simplest way may be to temporarily rename 
```
C:\Users\yijin\OneDrive\文档
```
to 
```
C:\Users\yijin\OneDrive\tempdir
```

SmartJingJing123, 2020-04-07:
@blthayer Yes, you are correct! After I renamed the folder to ASCII characters, all python versions passed the tests. I have tested python 3.4 3.6 3.7 3.8. So this script is good at my end now! Thanks for your reply!

blthayer, 2020-04-07:
@SmartJingJing123, great! I'm assuming that in the message above you meant "3.5" instead of "3.4"
@mzy2240 - as soon as you give this another shot we can close this issue.

mzy2240, 2020-04-07:
@blthayer I tested again, with all the suggestions you made, still got the same results. From the results I can tell all the dependencies are not properly installed, so lots of NameErrors are raised. Since I manually tested and passed all the versions, I suggest to close the issue.

mzy2240, 2020-04-07:
@blthayer The major NameError is _NameError: name 'saw' is not defined_ and I checked the terminal output and it showed everything was successfully installed.

blthayer, 2020-04-07:
@mzy2240 - that's quite odd... but I agree - since you've been able to manually test everything in all versions and since all passed for both @SmartJingJing123 and myself, I'll close the issue. I do think it's probably still worth investigating what's going on since ideally you would run the script prior to each release in order to automate the testing of different Python versions.

closed 2020-04-07 by blthayer

mzy2240, 2020-04-08:
@blthayer I managed to make the python 3.6 work with your code. I reinstalled the pywin32 using conda and then it is working now. But I still cannot make the py3.8 work even though I can manually pass the test in 3.8. The reason is largely due to the fact that the 3.8  venv may need to run the pywin32 post installation script. Do you know how to do that for a venv?

blthayer, 2020-04-08:
@mzy2240 - the pywin32 post-installation script just does some system-wide stuff. Can you post the output you're getting?

The next thing to try is to change the ``cmd_tuple`` that spans lines 69 and 70 from:
```python
        # Define command for installing ESA.
        cmd_tuple = \
            (exe, '-m', 'pip', 'install', '--upgrade', '--no-cache-dir')
```
to:
```python
        # Define command for installing ESA.
        cmd_tuple = \
            (exe, '-m', 'pip', 'install', '--upgrade', '--no-cache-dir', 
             '--only-binary', 'pywin32,pypiwin32')
```
I haven't tried it yet, it's possible that those last two items in the revised tuple should be one string instead of too, I'm not sure.

My personal advice: _never_ use Anaconda. It does too much stuff behind the scenes that lead to all these issues.

blthayer, 2020-04-10:
@mzy2240 - do you have a couple minutes to take a swing at this again? I just want to make sure we get the issues ironed out so that the script can be consistently used during the release process for quality control