mzy2240/ESA #38: Incorporate testing coverage into documentation

blthayer, 2020-04-02:
https://www.sphinx-doc.org/en/master/usage/extensions/coverage.html#sphinx.ext.coverage.coveragebuilder

blthayer, 2020-04-03:
@mzy2240 - this is complete. Please run `docs/rst/coverage_to_rst.py` after installing `coverage` and `tabulate` packages.

closed 2020-04-03 by blthayer

mzy2240, 2020-04-03:
@blthayer 
[image]
Is the output expected?

blthayer, 2020-04-04:
@mzy2240 - no, that's not expected. I'll try running with Python 3.8 and see if I get the same warning.

blthayer, 2020-04-04:
@mzy2240 - I was running from PyCharm, which does some automatic path patching. I've fixed it now, please try again.

This underscores the importance of running everything from the terminal during a release - we don't want the magic of PyCharm to hide things that are broken.

SmartJingJing123, 2020-04-04:
@blthayer I have some doubt after running the script. I got a result of Percentage Coverage = 100 in the README.rst, but there are some errors in the output, is this expected? Screenshot of the errors is attached:
[image]

blthayer, 2020-04-05:
@SmartJingJing123 - at the very beginning of the output, there's some text that says:

```
************************************************************************************************************************
RUNNING TESTS
Ignore any exceptions until you see a bunch of "*" characters.
```

The errors you see are intentionally raised during the tests - essentially testing that error handling is working appropriately. I went ahead and updated the printing in the script so that it's more clear that those printed errors are expected behavior. You can see my changes in the following commit: 7bcb2d3f839d4489a8edaed59389f4c5e5c30f60