mzy2240/ESA #43: Determine if we can add a "note" to citations for JOSS

blthayer, 2020-04-13:
A couple of our citations require notes (like "accepted, to be published after delayed conference takes place"). Brandon's thesis also needs a note since the full text won't be published for a while.

mzy2240, 2020-04-14:
@blthayer I managed to be able to add footnote. If you have not done this, I am happy to work on this. I did the same thing for the equal contribution footnote.

blthayer, 2020-04-14:
@mzy2240 - thank you! I'll look into this a bit tomorrow. What I'm going for, is with some citation managers you can add a "note" field to a BibTex entry. That field then shows up in the citation. I recently wrote a LaTex document where I added a "note" to my thesis indicating the full text won't be publicly available until 2022, and the note showed up in the references. However, with JOSS, it appears the "note" field doesn't render in the References. Does that make sense?

mzy2240, 2020-04-14:
Haha, I am actually looking for the same thing, but with no luck so far. So my solution is, adding the note directly in the markdown file, and it looks ok to me. Here is the snapshot:
[image]
But what I did for the equal contribution footnote involves more hack. What I did is to have a latex.template that can be used by the Pandoc and then it can auto-detect whether the author information has additional fields. I will update very soon once I verify the syle is the same as the JOSS uses.

blthayer, 2020-04-14:
@mzy2240 - did you find any JOSS documentation noting that we can tweak their template like that?

blthayer, 2020-04-14:
I'm going to close this for now, as @mzy2240 found a workable solution.

closed 2020-04-14 by blthayer