mzy2240/ESA #45: Release version 1.0.1 after all the merging and documentation dust has settled

blthayer, 2020-04-14:
After we've finished revising the documentation, merging branches, etc., let's release the next version of ESA. This adds some new features.

This brings up an important point: We should probably start keeping a change log/ version log where we note what's been added. I propose we add a section to the documentation and create a .rst file for tracking this.

Another important point: We need to start using `git tag` to add tags to the repository corresponding to our PyPi releases.

Both of the above "important points" should be documented in the `RELEASE_PROCESS.rst` file.

So I guess this issue covers all these points :) Let me know if you need any assistance, and please pull me in for review as needed.

I think this can be done either before or after the JOSS paper text is finalized, but should be done before we submit to JOSS (just to get things as clean as absolutely possible).

blthayer, 2020-04-14:
https://git-scm.com/book/en/v2/Git-Basics-Tagging

blthayer, 2020-04-15:
@mzy2240 - I'm going to assign this to you now. After you finish https://github.com/mzy2240/ESA/issues/31#issue-590608741 please go ahead and release version 1.0.1.

Please note that I've made significant updates to the release process file. Let me know if anything is confusing, wrong, or if you need a hand.

mzy2240, 2020-04-15:
@blthayer The new release process is very detailed! Great work! The only problem is I still have to manually change the version number in the __init__.py file, otherwise, it will still show version 1.0.0. I tried it in multiple environments (including brand new venv) and none of them work.

blthayer, 2020-04-15:
@mzy2240 - I'll file an issue and see if I can't figure it out. That's weird, I thought I had fixed that with my new regular expression. 

Thank you! Glad the process worked well otherwise.

closed 2020-04-15 by blthayer