mzy2240/ESA #46: Finalize JOSS paper, submit to JOSS

blthayer, 2020-04-15:
After receiving feedback from Professor's Overbye and Davis, make adjustments to the paper as needed. Then, submit to JOSS.

blthayer, 2020-04-16:
@mzy2240 @SmartJingJing123 - I think we're ready to submit after receiving the go-ahead from Professor Overbye.

blthayer, 2020-04-16:
Bad news: as of ~5 days ago, JOSS has suspended new submissions until May 4th.

https://joss.theoj.org/papers/new

mzy2240, 2020-04-17:
That is not good. I guess we have three options: wait till May 4th, try JORS or try NAPS. I personally prefer the first two options.

blthayer, 2020-04-17:
@mzy2240 @SmartJingJing123 - I propose we wait until May 4th. 

Does someone want to take point on putting together a NAPS paper in which we compare available simulators? We'd have to:

- Determine which simulators we have access to, and determine which free simulators to use
- Carefully design experiments using different cases and different scenarios. Perhaps we should use some cases/scenarios that were used for the GO competition?
- Literature review (has someone done this before)
- And more...

SmartJingJing123, 2020-04-17:
I propose to wait until May 4th as well. Since the paper is already written, we can wait for two more weeks, and then submit once JOSS submission is opened. NAPS is a choice but somewhat is time-consuming.

blthayer, 2020-04-17:
We'll wait until May 4th to submit and skip the NAPS paper based on Professor Overbye's feedback.

blthayer, 2020-06-17:
Paper has been submitted, and is near final acceptance:

https://github.com/openjournals/joss-reviews/issues/2289#issuecomment-645589393

closed 2020-06-17 by blthayer