mzy2240/ESA #47: Image fail to show up in the matplotlib example

mzy2240, 2020-04-15:
https://mzy2240.github.io/ESA/html/snippets.html#plot-histogram-of-line-flows-with-matplotlib

blthayer, 2020-04-15:
@mzy2240 @SmartJingJing123 - this is a bit weird. I pulled the latest from `develop`, and when I view the documentation locally (open it w/ Firefox through the Windows file system), the image shows up. 

So this may have to do with how the GitHub hosting of the documentation works. I see two things to try at first:

1. We could look into the Sphinx building options and see if there are different methods for including images
2. We could go back to @SmartJingJing123's original method and include the image via a web URL instead of a file path. This isn't my favorite option because it creates maintenance difficulties (local vs. remote changes), but if it works, it works.

Unfortunately, the only way to test these options is to make changes to `develop`, merge it into `master`, perform a `git push`, and then check the GitHub website. @mzy2240 - can you think of any better ways to double-check if it's working? Can we have a second documentation website for `develop`?

mzy2240, 2020-04-15:
Since the image is working now, the issue will be closed now.

closed 2020-04-15 by mzy2240