mzy2240/ESA #50: Publicize and/or advertise ESA

blthayer, 2020-04-17:
After we have an official publication listed as our citation, we should start advertising ESA. One example might include making a "Powerglobe" announcement.

blthayer, 2020-06-17:
@mzy2240, @SmartJingJing123 - I'll be emailing folks at PNNL about ESA now that it's "complete" and we've got the JOSS paper out. Would you guys be willing to take point on advertising it elsewhere?

SmartJingJing123, 2020-06-17:
https://github.com/mzy2240/ESA/issues/50#issuecomment-645636114
I could volunteer to advertise on Linkedin.

mzy2240, 2020-06-17:
Yeah I will advertise it to my intern company as well as friends in our field.

blthayer, 2020-06-24:
@mzy2240 @SmartJingJing123 - I've informed my PNNL network about ESA, and also updated the group regarding the JOSS publication. I propose we close this for now, and if anyone has more ideas for getting the word out, we can reopen.

blthayer, 2020-06-24:
@mzy2240 @SmartJingJing123 

We saw some decent cloning traffic before the JOSS paper came out as well as on the day the paper was published:

[image]

closed 2020-06-24 by blthayer