mzy2240/ESA #52: Remove the "modified" Texas 2000 bus case from the repository

blthayer, 2020-05-29:
The modified Texas 2000 bus case is huge (30MB), and we only use it for one test case: `ProcessAuxFileTestCase`. We should update that test case so that it does not depend on the modified case, and then delete the modified case from the repository.

While it may not seem like it's a big deal to have it around, the primary concern is that GitHub limits Git-LFS bandwidth for repositories unless you pay for more data. We should try to preemptively avoid this situation.

@mzy2240 - would you be willing to take this on after we get this next release out of the way?

mzy2240, 2020-05-29:
ok I will check whether I can use a smaller case to replace the current one

blthayer, 2020-05-29:
@mzy2240 - thank you! I only added that case because it had LTCs and I was using an LTC filter to test the ProcessAuxFile method... It's probably simplest to just create some new sort of aux file filter that works with something else within the 2000 bus case (e.g. generator fuel types) and rewrite the test accordingly.

closed 2020-06-02 by mzy2240