mzy2240/ESA #54: Add more tests to make the coverage 100%

mzy2240, 2020-05-29:
@blthayer Right now the coverage is about 96%. I checked the coverage reports and noticed the uncovered lines come from the newly-added exception handling for the old version. Since you are more familiar with mock module, can you add some tests for these exceptions?

blthayer, 2020-05-29:
@mzy2240 - I can do that, hopefully today, but I may not have time. I think it would be okay to release with 96% coverage and update the tests later if that's okay with you and I run out of time today.

mzy2240, 2020-05-29:
Okay no problem. Besides the coverage, it passed all the tests. I will release the new version later.

blthayer, 2020-05-29:
@mzy2240 - I was able to get this done, but won't have much more time for ESA in the next week.

Please note that one of the doctests is still failing because of the AreaNum issue.

closed 2020-05-29 by blthayer