mzy2240/ESA #56: Add support for decimal delimiters other than periods

blthayer, 2020-06-12:
In Europe, decimal numbers are delimited with a period, rather than a comma. Ensure that ESA seamlessly handles this difference.

This issue was brought up by `robinroche` in this [comment](https://github.com/openjournals/joss-reviews/issues/2289#issuecomment-643482550).

robinroche, 2020-06-14:
@blthayer Thanks for the fix but I still have the same issue.
The update seems to have worked:
```
(tutorial-env) C:\Users\rroche>python -m pip install --upgrade esa
Collecting esa
  Downloading esa-1.0.4-py3-none-any.whl (46 kB)
     |████████████████████████████████| 46 kB 421 kB/s
Requirement already satisfied, skipping upgrade: pywin32 in c:\users\rroche\tutorial-env\lib\site-packages (from esa) (227)
Requirement already satisfied, skipping upgrade: pypiwin32 in c:\users\rroche\tutorial-env\lib\site-packages (from esa) (223)
Requirement already satisfied, skipping upgrade: pandas>=0.25 in c:\users\rroche\tutorial-env\lib\site-packages (from esa) (1.0.4)
Requirement already satisfied, skipping upgrade: numpy>=1.13.3 in c:\users\rroche\tutorial-env\lib\site-packages (from esa) (1.18.5)
Requirement already satisfied, skipping upgrade: pytz>=2017.2 in c:\users\rroche\tutorial-env\lib\site-packages (from pandas>=0.25->esa) (2020.1)
Requirement already satisfied, skipping upgrade: python-dateutil>=2.6.1 in c:\users\rroche\tutorial-env\lib\site-packages (from pandas>=0.25->esa) (2.8.1)
Requirement already satisfied, skipping upgrade: six>=1.5 in c:\users\rroche\tutorial-env\lib\site-packages (from python-dateutil>=2.6.1->pandas>=0.25->esa) (1.15.0)
Installing collected packages: esa
  Attempting uninstall: esa
    Found existing installation: esa 1.0.3
    Uninstalling esa-1.0.3:
      Successfully uninstalled esa-1.0.3
Successfully installed esa-1.0.4
```
I get the following error (file test.py has not changed since my first comment):
```
(tutorial-env) C:\Users\rroche\Downloads>python test.py
21:12:23 [WARNING] [SAW]: UIVisible attribute could not be set. Note this SimAuto property was not introduced until Simulator version 20. Check your version with the get_simulator_version method.
Traceback (most recent call last):
  File "pandas\_libs\lib.pyx", line 1926, in pandas._libs.lib.maybe_convert_numeric
ValueError: Unable to parse string "  1,05999994"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 8, in <module>
    bus_data = saw.get_power_flow_results('bus')
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 465, in get_power_flow_results
    return self.GetParametersMultipleElement(ObjectType=object_type,
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 912, in GetParametersMultipleElement
    return self.clean_df_or_series(obj=df, ObjectType=ObjectType)
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 290, in clean_df_or_series
    obj[numeric_fields] = self._to_numeric(obj[numeric_fields])
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 1859, in _to_numeric
    return data.apply(lambda x: pd.to_numeric(x, errors=errors))
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\frame.py", line 6878, in apply
    return op.get_result()
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\apply.py", line 186, in get_result
    return self.apply_standard()
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\apply.py", line 313, in apply_standard
    results, res_index = self.apply_series_generator()
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\apply.py", line 341, in apply_series_generator
    results[i] = self.f(v)
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 1859, in <lambda>
    return data.apply(lambda x: pd.to_numeric(x, errors=errors))
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\tools\numeric.py", line 149, in to_numeric
    values = lib.maybe_convert_numeric(
  File "pandas\_libs\lib.pyx", line 1963, in pandas._libs.lib.maybe_convert_numeric
ValueError: Unable to parse string "  1,05999994" at position 0
```

blthayer, 2020-06-14:
@robinroche - thank you very much for working with me on this, and I'm sorry the initial fix wasn't adequate.

So that I can better reproduce the issue, could you please execute the following (using the `CASE_PATH` variable for your own machine) and provide me with the verbatim printed output?

```python
from esa import SAW
from esa.saw import convert_list_to_variant

saw = SAW(FileName=CASE_PATH)

# Check decimal delimiter.
print('Decimal delimiter:')
print(saw.decimal_delimiter)
print('')

saw.SolvePowerFlow()

# Call SAW internal functions for debugging purposes.
output = saw._call_simauto(
    'GetParametersMultipleElement', 'bus',
    convert_list_to_variant(['BusNum', 'BusName', 'BusPUVolt', 'BusAngle',
                             'BusNetMW', 'BusNetMVR']), '')
print('Results from GetParametersMultipleElement:')
print(output)
print('')

```

robinroche, 2020-06-14:
@blthayer Sure, here is the output:

```
(tutorial-env) C:\Users\rroche\Downloads>python test2.py
21:54:21 [WARNING] [SAW]: UIVisible attribute could not be set. Note this SimAuto property was not introduced until Simulator version 20. Check your version with the get_simulator_version method.
Decimal delimiter:
.

Results from GetParametersMultipleElement:
(('    1', '    2', '    3', '    4', '    5', '    6', '    7', '    8', '    9', '   10', '   11', '   12', '   13', '   14'), ('Bus 1', 'Bus 2', 'Bus 3', 'Bus 4', 'Bus 5', 'Bus 6', 'Bus 7', 'Bus 8', 'Bus 9', 'Bus 10', 'Bus 11', 'Bus 12', 'Bus 13', 'Bus 14'), ('  1,05999994', '  1,04499996', '  1,00999999', '  1,01767162', '  1,01951459', '  1,07000005', '  1,06152019', '  1,09000003', '  1,05593281', '  1,05098559', '  1,05690694', '  1,05518907', '  1,05038265', '  1,03553058'), ('  0,00000000', ' -4,98255334', '-12,72502699', '-10,31282882', ' -8,77379918', '-14,22086891', '-13,35955842', '-13,35957101', '-14,93845750', '-15,09722071', '-14,79055172', '-15,07551240', '-15,15619553', '-16,03356498'), ('232,39169121', ' 18,30000132', '-94,19999719', '-47,79999852', ' -7,59999976', '-11,20000035', '  0,00000000', '  0,00000000', '-29,49999869', ' -9,00000036', ' -3,50000001', ' -6,10000007', '-13,50000054', '-14,90000039'), ('-16,54938906', ' 30,85595667', '  6,07485175', '  3,90000008', ' -1,60000008', '  5,22969961', '  0,00000000', ' 17,62306690', '  4,58488762', ' -5,79999983', ' -1,79999992', ' -1,60000008', ' -5,79999983', ' -5,00000007'))
```

blthayer, 2020-06-14:
@robinroche - thanks for your rapid response. 

It's odd to me that it's reporting your decimal delimiter as a period, when it's clearly coming back from PowerWorld as a comma. This is my first time using Python's `locale` module, so I may be misunderstanding it.

Can you execute the following code and provide me with the output?

```python
import locale
print(locale.getdefaultlocale())
```

robinroche, 2020-06-14:
@blthayer Here is what I get:
```
(tutorial-env) C:\Users\rroche\Downloads>python
Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:20:19) [MSC v.1925 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import locale
>>> print(locale.getdefaultlocale())
('fr_FR', 'cp1252')
```

blthayer, 2020-06-14:
@robinroche, that's very helpful, thank you. I _think_ I know how to fix this now.

To confirm, can you please try the following two things:
```python
import locale
locale.setlocale(locale.LC_NUMERIC, '')
d = locale.localeconv()
print(d['decimal_point'])
```
and
```python
import locale
locale.setlocale(locale.LC_NUMERIC, 'fr_FR')
d = locale.localeconv()
print(d['decimal_point'])
```

robinroche, 2020-06-14:
@blthayer Here is the output:
```
(tutorial-env) C:\Users\rroche\Downloads>python test3.py
,

(tutorial-env) C:\Users\rroche\Downloads>python test4.py
Traceback (most recent call last):
  File "test4.py", line 2, in <module>
    locale.setlocale(locale.LC_NUMERIC, 'fr_FR')
  File "C:\Users\rroche\AppData\Local\Programs\Python\Python38-32\lib\locale.py", line 608, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

blthayer, 2020-06-14:
@robinroche - thank you. I am now quite sure I've figured out the correct way to fix this. 

In my initial fix, I incorrectly assumed that the `locale` module automatically uses your current locale. In fact, you first have to set the locale, and the line `locale.setlocale(locale.LC_ALL, '')` sets all your locale information from your system.

I'll be able to push a fix to this right away, and hopefully @mzy2240 will be able to provide us with a new release for you to try shortly.

closed 2020-06-17 by blthayer