mzy2240/ESA #61: Add Lines to Case example not working for all PWS versions or all locales

blthayer, 2020-06-15:
_Originally posted by @robinroche in https://github.com/openjournals/joss-reviews/issues/2289#issuecomment-644265540_

I'm now testing the "Add Lines to Case" example.
I'm getting the following:
```
(tutorial-env) C:\Users\rroche\Downloads>python test5.py
   From Number  To Number  Ckt        R        X        B  Lim MVA A
0         8155       5358    3  0.00037  0.00750  0.52342       2768
1         8154       8135    3  0.00895  0.03991  0.00585        149
2         8153       8108    3  0.01300  0.05400  0.02700        186
3         8152       8160    3  0.00538  0.03751  0.00613        221
4         8155       8057    3  0.00037  0.00750  0.52342       2768
5         8154       8153    3  0.01300  0.05400  0.02700        186
6         8155       8135    3  0.00538  0.03751  0.00613        221
19:18:30 [WARNING] [SAW]: UIVisible attribute could not be set. Note this SimAuto property was not introduced until Simulator version 20. Check your version with the get_simulator_version method.
Index(['BusNum', 'BusNum:1', 'LineCircuit', 'LineR', 'LineX', 'LineC',
       'LineAMVA'],
      dtype='object')
['BusNum', 'BusNum:1', 'LineCircuit']
Traceback (most recent call last):
  File "test5.py", line 34, in <module>
    saw.change_and_confirm_params_multiple_element('branch', line_df)
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 202, in change_and_confirm_params_multiple_element
    cleaned_df = self._change_parameters_multiple_element_df(
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 1754, in _change_parameters_multiple_element_df
    cleaned_df = self.clean_df_or_series(obj=command_df,
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 290, in clean_df_or_series
    obj[numeric_fields] = self._to_numeric(obj[numeric_fields])
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 1852, in _to_numeric
    data = data.apply(lambda x:
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\frame.py", line 6878, in apply
    return op.get_result()
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\apply.py", line 186, in get_result
    return self.apply_standard()
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\apply.py", line 295, in apply_standard
    result = libreduction.compute_reduction(
  File "pandas\_libs\reduction.pyx", line 620, in pandas._libs.reduction.compute_reduction
  File "pandas\_libs\reduction.pyx", line 128, in pandas._libs.reduction.Reducer.get_result
  File "C:\Users\rroche\tutorial-env\lib\site-packages\esa\saw.py", line 1853, in <lambda>
    x.str.replace(self.decimal_delimiter, '.'))
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\generic.py", line 5270, in __getattr__
    return object.__getattribute__(self, name)
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\accessor.py", line 187, in __get__
    accessor_obj = self._accessor(obj)
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\strings.py", line 2041, in __init__
    self._inferred_dtype = self._validate(data)
  File "C:\Users\rroche\tutorial-env\lib\site-packages\pandas\core\strings.py", line 2098, in _validate
    raise AttributeError("Can only use .str accessor with string values!")
AttributeError: Can only use .str accessor with string values!
```
Lines 33-34 in the script are:
```
(33) saw.RunScriptCommand("EnterMode(EDIT);")
(34) saw.change_and_confirm_params_multiple_element('branch', line_df)
```
I don't know if this is another locale issue, so just in case, I tried adding the following to the script, but the result remains the same:
```
import locale
locale.setlocale(locale.LC_ALL, '')
```

blthayer, 2020-06-15:
@mzy2240 - I'm busy today, so if you wanted to look into this, you're welcome to. Otherwise, I'll try to tackle it this evening. This could either be an issue with using PWS 17, or a locale issue (or something else entirely!). The example works for me with PWS 21.

We'll likely need to pinpoint where things are going wrong, and ask `robinroche` to run some internal `SAW` methods and give us the output so we can put it into a test case. See `test_get_power_flow_results_with_commas` for an example of how I've done this.

Also, please note that the aforementioned test method is not present on the master branch. So, something went wrong with your merging. Do you want me to file a separate ticket for that?

mzy2240, 2020-06-15:
Let me check what's missing in the master branch and try to solve from my end. If I cannot, I will let you know.

blthayer, 2020-06-15:
@mzy2240 - thank you! I think I have a hunch for how to fix the issue in this ticket. I'm happy to fix it myself this evening, or you're welcome to take it on.

In short, my new `_to_numeric` method assumes that all column data types are `object` (I think that's what Numpy/Pandas calls it). In other words, it assumes all columns are non-numeric.

To fix, we just need to adapt the `_to_numeric` method to check column data types in the beginning, and only operate on non-numeric columns.

mzy2240, 2020-06-15:
@blthayer The missing part is back. And for the future convenience, I merged the master back to develop to make the develop branch even with the master branch (before this, the commit history are different, while the files are the same).

blthayer, 2020-06-15:
@mzy2240 - thank you! I'll take a swing at fixing this issue this evening, and if I need additional help I'll let you know. If you're willing to help me push out yet another release in a short period of time, I'd appreciate it. Sorry for all the back and forth here - these issues have been unexpected and difficult to test for since they're related to old PWS versions and locale issues.

mzy2240, 2020-06-15:
Yeah absolutely not a problem!

mzy2240, 2020-06-16:
The new version is released.

closed 2020-06-17 by blthayer