mzy2240/ESA #64: Create final release for JOSS

blthayer, 2020-06-17:
@mzy2240 - could you please create a documentation-only release? No code has changed since `v1.0.6` was released, so I propose doing the following:

- Merge `master` into `develop` to incorporate commits from `danielskatz`
- Delete the `v1.0.6` tag both locally and remotely ([link](https://devconnected.com/how-to-delete-local-and-remote-tags-on-git/))
- Recreate the `v1.0.6` tag, pointing at the current tip of `master`
- Update `paper.pdf`? This is optional in my view.
- Do we need to replace the `v1.0.6` release on PyPi? I don't think those releases include the documentation, right? If the releases do not include the documentation, I don't think we need to do anything with respect to PyPi

Anything else? Please see this comment:
https://github.com/openjournals/joss-reviews/issues/2289#issuecomment-645538830

I'll then upload an archive of the software at v1.0.6 to Zenodo.

mzy2240, 2020-06-17:
@blthayer From my understanding, it is not possible to update the PYPI release with same version number.

blthayer, 2020-06-17:
@mzy2240 - does the PyPi release have any of the documentation packaged with it? I don't think it does, but would need to double-check/investigate. If it doesn't, I don't think we need to change anything on the PyPi side.

mzy2240, 2020-06-17:
It doesn't. It only covers the README in the ROOT directory.

mzy2240, 2020-06-17:
I just remove the old tag and re-create the new tag for v1.0.6.
Can you update the paper.pdf file? I think that is the only remaining part for this issue. Or just let me know where I can get the latest pdf file, I can update it also.

blthayer, 2020-06-17:
For this new tag, let's use the following (if you could upload it, I'd appreciate it):

https://github.com/openjournals/joss-reviews/issues/2289#issuecomment-645529505

After the paper is published, we'll update with the final version.

mzy2240, 2020-06-17:
I just uploaded the latest pdf with an associated tag. Please go ahead to upload to Zenodo.

closed 2020-06-17 by blthayer