mzy2240/ESA #67: Update all citation locations with the new JOSS citation after published

blthayer, 2020-06-17:


mzy2240, 2020-06-17:
Just modified my N-X paper with the latest citation.
@SmartJingJing123 Can you check whether the ref in the RIDES-related papers need to update also?

blthayer, 2020-06-17:
@mzy2240 - I'm going to close this since I've updated the citation in the main README as well as in the .rst file. I'll go ahead and email the group with the new citation.

closed 2020-06-17 by blthayer

SmartJingJing123, 2020-06-17:
> Just modified my N-X paper with the latest citation.
> @SmartJingJing123 Can you check whether the ref in the RIDES-related papers need to update also?

Yes, I will add ESA in the evaluation metric paper since I used it in the TSR file decoding process. And I could think about whether ESA can be added in the RIDES development paper as well.