mzy2240/ESA #76: installing esa via Cygwin

faceb912, 2022-05-10:
Hi all, 
Can I install esa inside Cygwin ? i tried but i faced some issues !

mzy2240, 2022-05-10:
Hi @faceb912, I don't have any experience with Cygwin. Have you tried installing with pip?

faceb912, 2022-05-12:
> Hi @faceb912, I don't have any experience with Cygwin. Have you tried installing with pip?

Hi Mzy, It works fine with with pip. However, when i tried to install it inside the Cygwin using pip , it shows that pywin32  is missing. when i tried to install it pypwin32  first , i got this error 
ERROR: Could not find a version that satisfies the requirement pywin32 (from versions: none)
ERROR: No matching distribution found for pywin32

faceb912, 2022-05-19:
> Here's my understanding:
> 
> PowerWorld is a Windows only software tool. It utilizes the Window's [Component Object Model (COM)](https://docs.microsoft.com/en-us/windows/win32/com/the-component-object-model) for I/O with external programs/tools like `esa`.
> 
> `esa` uses [pywin32](https://github.com/mhammond/pywin32) in order to access PowerWorld's COM server interface.
> 
> The error you should ("could not find a version...") indicates that `pywin32` doesn't work under Cygwin out of the box, and therefore neither does `esa`.
> 
> I would have to do significantly more research on how Cygwin, COM, and `pywin32` work in order to give a more detailed answer. You could attempt to build [pywin32 from source](https://github.com/mhammond/pywin32#building-from-source), but I don't know if that'll actually work. It might work if Cygwin allows access to COM 🤷

Thank u for ur suggestions. I would try to build it from the source and I will let u know then

mzy2240, 2022-05-21:
Feel free to re-open the ticket if you still have questions.

closed 2022-05-21 by mzy2240

blthayer, 2022-10-11:
I would suspect that installation via Cygwin would not work, but I would
need to look into it  further to give a 100% confident answer. PowerWorld
is Windows-only and ESA utilizes PowerWorld's COM interface for
data/command I/O. COM is a special Windows-only feature that I suspect (but
have not confirmed) cannot be used from Cygwin.