mzy2240/ESA #77: Issue when call change multiple parameters function

faceb912, 2022-05-26:
Hi All , 
I have tried to follow the example in the website. However, I got error when I tried to change the active power in Gen. any suggestions? 
`C:\Users\Mike\Desktop\jen>python se.py
   BusNum GenID       GenMW     GenMVR
0       1     1  232.391691 -16.549389
1       2     1   40.000001  43.555957
2       3     1    0.000000  25.074852
3       6     1    0.000000  12.729700
4       8     1    0.000000  17.623067
12:05:04 [ERROR] [SAW]: An error occurred when trying to call ChangeParametersMultipleElement with ('gen', win32com.client.VARIANT(8204, ['BusNum', 'GenID', 'GenMW']), [win32com.client.VARIANT(8204, 3), win32com.client.VARIANT(8204, '1'), win32com.client.VARIANT(8204, 30)])
Traceback (most recent call last):
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python39\lib\site-packages\esa\saw.py", line 2409, in _call_simauto
    output = f(*args)
  File "<COMObject pwrworld.SimulatorAuto>", line 2, in ChangeParametersMultipleElement
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python39\lib\site-packages\win32com\client\dynamic.py", line 368, in _ApplyTypes_
    result = self._oleobj_.InvokeTypes(
TypeError: Objects for SAFEARRAYS must be sequences (of sequences), or a buffer object.
Traceback (most recent call last):
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python39\lib\site-packages\esa\saw.py", line 2409, in _call_simauto
    output = f(*args)
  File "<COMObject pwrworld.SimulatorAuto>", line 2, in ChangeParametersMultipleElement
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python39\lib\site-packages\win32com\client\dynamic.py", line 368, in _ApplyTypes_
    result = self._oleobj_.InvokeTypes(
TypeError: Objects for SAFEARRAYS must be sequences (of sequences), or a buffer object.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Mike\Desktop\ESA\se.py", line 10, in <module>
    saw.ChangeParametersMultipleElement(ObjectType='gen', ParamList=params, ValueList=values)
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python39\lib\site-packages\esa\saw.py", line 1381, in ChangeParametersMultipleElement
    return self._call_simauto('ChangeParametersMultipleElement',
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python39\lib\site-packages\esa\saw.py", line 2414, in _call_simauto
    raise COMError(m)
esa.saw.COMError: An error occurred when trying to call ChangeParametersMultipleElement with ('gen', win32com.client.VARIANT(8204, ['BusNum', 'GenID', 'GenMW']), [win32com.client.VARIANT(8204, 3), win32com.client.VARIANT(8204, '1'), win32com.client.VARIANT(8204, 30)])`

mzy2240, 2022-05-26:
Hi @faceb912, there is only very limited information you gave about the issue, so here is my best guess:

* Check your `ChangeParametersMultipleElement` function and make sure you are giving the right parameters. If you are following the example, please accommodate based on the case you are running.
* For most users, the helper functions like `change_and_confirm_params_multiple_element` and `change_parameters_multiple_element_df` are generally more pythonic and straightforward for changing simulation parameters. Here is an example that could help you start: https://mzy2240.github.io/ESA/html/snippets.html#increase-loading-in-case.

mzy2240, 2022-06-03:
@faceb912 Feel free to re-open the ticket if you still encounter the same issue.

closed 2022-06-03 by mzy2240