mzy2240/ESA #78: ValueError when attempting to access bus Marginal Cost.

Weifarers, 2022-06-23:
The following code operates as expected for a simple 37 bus system. 

```python
saw = esa.SAW(FileName=case_path)

saw.SolvePowerFlow()

bus_key_fields = saw.get_key_field_list('bus')

base_bus_results = saw.GetParametersMultipleElement('bus', bus_key_fields + ['BusPUVolt'])
```

This outputs: 
```
    BusNum  BusPUVolt
0        1   1.012034
1        3   0.966774
2        5   0.998733
3       10   1.021065
4       12   1.026115
5       13   0.984075
6       14   0.970844
7       15   0.984631
8       16   0.986146
9       17   0.999299
10      18   1.005596
...
```
However, I am also interested in looking for the marginal cost for the various buses in the system. Upon trying to execute the command:

```python
saw.GetParametersMultipleElement('bus', bus_key_fields + ['BusMCMW'])
```

where `BusMCMW` is listed at `https://github.com/mzy2240/ESA/blob/master/docs/power_world_object_fields.xlsx` as the Variable Name for the marginal cost, I get the following error.

`ValueError: Unable to parse string "Not Valid" at position 0`

Trying the command with the Concise Variable Name
```python
saw.GetParametersMultipleElement('bus', bus_key_fields + ['MargCostMW'])
```

returns the error 

`ValueError: The given object has fields which do not match a PowerWorld internal field name!`

Although I think this is an expected behavior, looking at #1 . So I'm not entirely sure what is going on with my usage of `BusMCMW`.

mzy2240, 2022-06-23:
I am able to reproduce the described error and I think this is an expected (and correct) result. The major problem of this use case is the lack of OPF results in the case. I did check the same case in the PW simulator, and in the model explorer for the BusMCMW column it also shows "Not Valid". So my suggestion is, try to explicitly run the OPF before querying the LMP values.

Weifarers, 2022-06-23:
Got it; running the command:

```python 
saw.RunScriptCommand("SolvePrimalLP")
```

before accessing the `BusMCMW` seems to have worked. 

Interestingly, I tried this with a different case (using the same code), and it seems that I could access `BusMCMW` after only initializing the SAW object (that is, before I even ran something like 
```python 
saw.SolvePowerFlow()
```
or
```python 
saw.RunScriptCommand("SolvePrimalLP")
```

I assume that this is a behavior that is dependent on the case, and is similar to prior discussions, where it is better to ensure the results are there by running an OPF first, which guarantees the expected values we want will be there?

mzy2240, 2022-06-23:
Feel free to re-open the issue if you still have questions.

closed 2022-06-23 by mzy2240