mzy2240/ESA #98: get_lodf() function doesn't work properly with open lines

MaxB25, 2023-03-16:
When doing contingency analysis, and some lines are open, the lodf matrix that is returned is incorrect.

mzy2240, 2023-03-16:
`get_lodf_matrix` or `get_lodf_matrix_fast`?

MaxB25, 2023-03-17:
It seems to be both of them, get_lodf_matrix returns the correct values but out of place (usually everything is just shifted one column to the left), and get_lodf_matrix_fast gets a singular matrix error when lines are out of service.

mzy2240, 2023-03-17:
Thanks for the heads-up! I think ‘get_lodf_matrix’ should be an easy fix, as long as we figured out the pattern of change in the PW GUI when there is line outage. For ‘get_lodf_matrix_fast’ it will be more tricky, cause we might need to fix the injection shift factor calculation under line outage first. @SanjanaKunkolienkar @SmartJingJing123 can you take a look at ‘get_lodf_matrix’ and let me know your thoughts? We probably just need to scan the branch table and then reindexing the lodf matrix, or creating a lodf dataframe with a human-readable column header.

mzy2240, 2023-06-05:
@MaxB25 1.3.5 is released and the issue should be fixed. Thanks for your patience!

closed 2024-03-06 by mzy2240