mzy2240/ESA #105: Error with # branch greater than 1000 for "run_contingency_analysis"

weigao-123, 2023-05-31:
Code for duplicating the error:
`python
    from esa import SAW
    case = r'tx2000_base_pws_version_22.pwb'
    saw = SAW(case)
    saw.SolvePowerFlow()
    saw.run_contingency_analysis('N-1')
`

Error message:
`Traceback (most recent call last):
  File "...", line 756, in <module>
    saw.run_contingency_analysis('N-1')
  File "...\lib\site-packages\esa\saw.py", line 1305, in run_contingency_analysis
    c1_isl, count, self.lodf, f, lim)
  File "...\lib\site-packages\esa\saw.py", line 1701, in n1_fast
    violations[violating_lines] += 1
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed`

The error is caused by the numpy indexing, when the size of the branch is greater than 1000, we get "np.matrix" type for "violating_lines" instead of "np.array" type. So we could solve this issue by:

- violations[violating_lines] += 1
+ violations[np.squeeze(np.asarray(violating_lines))] += 1

I can also create a PR if you need

mzy2240, 2023-05-31:
That's quite surprising consider we have tested this implementation many many times on the synthetic 2000 bus case. Do you mind to share your numpy, pandas, and esa version?

weigao-123, 2023-05-31:
Well, actually I did not find it in the tests folder, but here it is:
```python
import pandas as pd
import esa
print(np.__version__)
print(pd.__version__)
print(esa.__version__)
1.21.5
1.3.5
1.3.4
```

mzy2240, 2023-06-01:
Looks fine. Can you please go ahead to submit a PR? Let's see if the test would be happy.

cuihantao, 2023-06-01:
`numpy.matrix` is getting deprecated. Might just use `np.ravel(violating_lines)` if `violating_lines` is an array-like.

weigao-123, 2023-06-01:
@cuihantao that is good to know, more concise, thank you!

closed 2023-06-01 by mzy2240