mzy2240/ESA #109: Inconsistencies with Uniform Load Scaling and Power Flow convergence?

Weifarers, 2023-06-03:
I am attempting to automate a process in which I outage a set of generators for a case where I know it does not converge for the base case, and uniformly scale the load to check if it still converges. However, I'm running into an inconsistency that I did not anticipate: 

```python
    # Opening the case. Note: ESA requires an absolute path, not a relative one.
    case_path = r'C:\Users\trinh7\OneDrive - LLNL\Documents\White Walker\Consequence Enumeration' \
                r'\ESA Test\10k Case\ACTIVSg10k.pwb'

    # Given the case path, we first generate the SAW object.
    saw = SAW(FileName=case_path)

    # Then we solve the power flow, and save the initial state.
    saw.SolvePowerFlow()

    # We then want all the relevant parameters for generators. The SAW command gets us the most important numerical
    # identifiers for generators, but then we augment it with specific fields that we're interested in.
    gen_key_fields = saw.get_key_field_list('gen')
    gen_params = gen_key_fields + ['BusName', 'GenMW', 'GenStatus', 'BusCat']
    gen_data = saw.GetParametersMultipleElement(ObjectType='gen', ParamList=gen_params)

    # Then we consider only generators that are already closed.
    gen_data = gen_data[gen_data['GenStatus'] == 'Closed']

    # Also, we remove the slack bus from consideration.
    gen_data = gen_data[gen_data['BusCat'] != 'Slack']

    # Because generators are mapped using a bus number and generator ID, we zip this information together into a
    # list, and append it to the end of the DataFrame. This makes it easier to reference and outage exactly the
    # generators we want to, especially when there are multiple generators at one bus, so you can't just rely on
    # the bus number. We also use the bus name, just in case as well.
    gen_data['list'] = [list(a) for a in zip(gen_data['BusNum'], gen_data['GenID'], gen_data['BusName'])]

    # We also need the relevant parameters and information for the loads, since we'll be uniformly scaling the load
    # in order to reach convergence.
    load_key_fields = saw.get_key_field_list('load')
    load_params = load_key_fields + ['LoadSMW', 'LoadSMVR']
    load_data = saw.GetParametersMultipleElement(ObjectType='load', ParamList=load_params)

    # Defining the generators we want to outage.
    outage_list = [[40843, '1', 'PHOENIX 74 4'], [40844, '1', 'PHOENIX 74 5'], [10811, '1', 'RICHLAND 4 4'],
                   [26134, '1', 'AVILA BEACH 5'], [26133, '1', 'AVILA BEACH 4'], [11016, '1', 'GRAND COULEE 6'],
                   [11018, '1', 'GRAND COULEE 8'], [50386, '1', 'DELTA 2 4'], [50385, '1', 'DELTA 2 3'],
                   [10684, '1', 'MATTAWA 2 5']]

    # We create a copy of the generator information to modify and pass back to PowerWorld.
    gen_copy = gen_data.copy(deep=True)
    # Then we replace all the generator information so that the generators will be open instead of closed.
    gen_copy.loc[(gen_copy['list'].isin(outage_list)), 'GenStatus'] = ['Open'] * len(outage_list)

    # We then change and confirm the parameter change. Note that command_df requires that the dataframe only
    # contains the key fields and the parameters that are getting changed, so we drop everything else.
    saw.change_and_confirm_params_multiple_element(ObjectType='gen',
                                                   command_df=gen_copy.drop(
                                                       ['GenMW', 'BusName', 'BusCat', 'list'],
                                                       axis=1))
    saw.SaveState()

    # We create a list of load multipliers to test.
    load_multiplier_list = [99, 99.9, 99.99, 99.999]

    print('Load Multiplier List Order: ', load_multiplier_list)
    for load_multiplier in load_multiplier_list:
        # We generate a copy of this data for re-loading purposes.
        load_copy = load_data.copy(deep=True)

        # Then we try to solve the power flow. If the case converges, we take note of that.
        try:
            load_copy[['LoadSMW', 'LoadSMVR']] *= (load_multiplier / 100)
            saw.change_parameters_multiple_element_df(ObjectType='load', command_df=load_copy)
            saw.SolvePowerFlow()
            print("Successfully solved case with Load Multiplier of {}%"
[... 104 lines cut ...]
    print('----------------------------------------------------------------')

    saw.exit()
```

This returns:
```
Inverse Load Multiplier List Order:  [99.999, 99.99, 99.9, 99]
Unsuccessfully solved case with Load Multiplier of 99.999%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Unsuccessfully solved case with Load Multiplier of 99%
----------------------------------------------------------------
```

If I limit the list to just be `load_multiplier_list = [99]`, both functions return that the case successfully solved, so I'm not sure what's going on here. Is there a more appropriate way of modifying the case to do this sort of testing? Am I missing something in the logic chain that causes this issue?

For further testing, I also attempted to end each loop with: 
```python
# At the end of every loop, we then exit the file, then re-open it and re-solve the initial power flow.
saw.exit()
saw = SAW(FileName=case_path)
saw.SolvePowerFlow()
```
and observed similar results. I also attempted to move the `saw.SaveState()` function further up, right after I solved the initial power flow and before I opened any generators (although it's mentioned in the [documentation](https://mzy2240.github.io/ESA/html/esa.html?highlight=loadstate) that `saw.LoadState()` doesn't function properly if the system topology is modified, so observing inconsistencies there seems to be expected).

mzy2240, 2023-06-03:
can you try passing `pw_order=True` when initialize the SAW object? this will always give you the order that PW is using internally, which I suspect is the reason for the inconsistency. The side effect of setting pw_order to True is you have to take care of the dataframe column data type yourself.

Weifarers, 2023-06-03:
Still encountering issues, with modified code:

```python
def save_load_post_gen():
    # Opening the case. Note: ESA requires an absolute path, not a relative one.
    case_path = r'path\to\ACTIVSg10k.pwb'

    # Given the case path, we first generate the SAW object.
    saw = SAW(FileName=case_path, pw_order=True)

    # Then we solve the power flow, and save the initial state.
    saw.SolvePowerFlow()

    # We then want all the relevant parameters for generators. The SAW command gets us the most important numerical
    # identifiers for generators, but then we augment it with specific fields that we're interested in.
    gen_key_fields = saw.get_key_field_list('gen')
    gen_params = gen_key_fields + ['BusName', 'GenMW', 'GenStatus', 'BusCat']
    gen_data = saw.GetParametersMultipleElement(ObjectType='gen', ParamList=gen_params)

    # Then we consider only generators that are already closed.
    gen_data = gen_data[gen_data['GenStatus'] == 'Closed']

    # Also, we remove the slack bus from consideration.
    gen_data = gen_data[gen_data['BusCat'] != 'Slack']

    # Because generators are mapped using a bus number and generator ID, we zip this information together into a
    # list, and append it to the end of the DataFrame. This makes it easier to reference and outage exactly the
    # generators we want to, especially when there are multiple generators at one bus, so you can't just rely on
    # the bus number. We also use the bus name, just in case as well.
    gen_data['list'] = [list(a) for a in zip(gen_data['BusNum'], gen_data['GenID'], gen_data['BusName'])]

    # We also need the relevant parameters and information for the loads, since we'll be uniformly scaling the load
    # in order to reach convergence.
    load_key_fields = saw.get_key_field_list('load')
    load_params = load_key_fields + ['LoadSMW', 'LoadSMVR']
    load_data = saw.GetParametersMultipleElement(ObjectType='load', ParamList=load_params)

    # Defining the generators we want to outage.
    outage_list = [[' 40843', '1', 'PHOENIX 74 4'], [' 40844', '1', 'PHOENIX 74 5'], [' 10811', '1', 'RICHLAND 4 4'],
                   [' 26134', '1', 'AVILA BEACH 5'], [' 26133', '1', 'AVILA BEACH 4'], [' 11016', '1', 'GRAND COULEE 6'],
                   [' 11018', '1', 'GRAND COULEE 8'], [' 50386', '1', 'DELTA 2 4'], [' 50385', '1', 'DELTA 2 3'],
                   [' 10684', '1', 'MATTAWA 2 5']]

    # We create a copy of the generator information to modify and pass back to PowerWorld.
    gen_copy = gen_data.copy(deep=True)
    # Then we replace all the generator information so that the generators will be open instead of closed.
    gen_copy.loc[(gen_copy['list'].isin(outage_list)), 'GenStatus'] = ['Open'] * len(outage_list)

    # We then change and confirm the parameter change. Note that command_df requires that the dataframe only
    # contains the key fields and the parameters that are getting changed, so we drop everything else.
    saw.change_and_confirm_params_multiple_element(ObjectType='gen',
                                                   command_df=gen_copy.drop(
                                                       ['GenMW', 'BusName', 'BusCat', 'list'],
                                                       axis=1))
    saw.SaveState()

    # We create a list of load multipliers to test.
    load_multiplier_list = [99, 99.9, 99.99, 99.999]

    print('Load Multiplier List Order: ', load_multiplier_list)
    for load_multiplier in load_multiplier_list:
        # We generate a copy of this data for re-loading purposes.
        load_copy = load_data.copy(deep=True)

        # Then we try to solve the power flow. If the case converges, we take note of that.
        try:
            load_copy['LoadSMW'] = [float(load_smw) for load_smw in load_copy['LoadSMW']]
            load_copy['LoadSMVR'] = [float(load_smvr) for load_smvr in load_copy['LoadSMVR']]
            load_copy[['LoadSMW', 'LoadSMVR']] *= (load_multiplier / 100)
            saw.change_parameters_multiple_element_df(ObjectType='load', command_df=load_copy)
[... 92 lines cut ...]

        # At the end of every loop, we then re-load the original state of the system.
        saw.LoadState()
    print('----------------------------------------------------------------')

    saw.exit()
```

Running these both return:
```
Load Multiplier List Order:  [99, 99.9, 99.99, 99.999]
Successfully solved case with Load Multiplier of 99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.999%
----------------------------------------------------------------
Inverse Load Multiplier List Order:  [99.999, 99.99, 99.9, 99]
Unsuccessfully solved case with Load Multiplier of 99.999%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Unsuccessfully solved case with Load Multiplier of 99%
----------------------------------------------------------------

Process finished with exit code 0
```

mzy2240, 2023-06-03:
might caused by power flow options, try reset to flat start for both.

Weifarers, 2023-06-03:
Just to confirm, resetting to flat start would involve replacing `saw.LoadState()` with `saw.RunScriptCommand2(Statements='ResetToFlatStart', StatusMessage='fail')`, correct? 

If that is the case, I did so and found the same inconsistencies.

mzy2240, 2023-06-03:
Have you compared the state after restoring the state and resetting to flat start?

Weifarers, 2023-06-03:
Double-checking what you mean by that. I compared the same code, except at the end of each loop I had:

```python
saw.LoadState()
print(saw.get_power_flow_results('bus')['BusPUVolt'])
```

versus:

```python
saw.LoadState()
saw.RunScriptCommand2(Statements='ResetToFlatStart', StatusMessage='true')
print(saw.get_power_flow_results('bus')['BusPUVolt'])
```

This returns what you would expect, which is that resetting to flat start does set the majority of bus voltages to 1.0 pu. 

Is resetting to flat start the expected procedure? Even opening a fresh 10k system case from the electric grids repository, if I reset the system to flat start and then attempt to solve the power flow, the system doesn't converge.

mzy2240, 2023-06-03:
Probably you want to always start from a feasible solution. Save the 99% solution and always start from there might be a reasonable solution.

Weifarers, 2023-06-03:
When you say start from feasible solutions, do you mean starting the analysis? The start of the code
```python
    # Opening the case. Note: ESA requires an absolute path, not a relative one.
    case_path = r'path\to\ACTIVSg10k.pwb'

    # Given the case path, we first generate the SAW object.
    saw = SAW(FileName=case_path, pw_order=True)

    # Then we solve the power flow.
    saw.SolvePowerFlow()
```

Is my attempt at starting the analysis from a feasible solution; while this specific combination of generators converges at 99%, I anticipate other cases where it does not converge at 99% (and I want my code to accurately reflect that). 

Fundamentally, I want to replicate a manual process where I open the case, outage a set of generators and uniformly change the load, and then solve the power flow. 

The expected result (from my manual testing, at least), is that the system converges at 99% loading, but then doesn't converge at 99.9%, 99.99%, or 99.999%. (This is the result you observe when you have `load_multiplier_list = [99, 99.9, 99.99, 99.999]`). But I don't see why inverting the order of the list changes the results of a power flow solve. 

I guess this brings up the other question of what's the best way to iteratively test a variety of changes on a system? My assumption was that `saw.SaveState()` and `saw.LoadState()` were the preferred commands (as it's mentioned in the documentation), but from my testing I'm not observing consistency in results, unless it's the fact that I'm changing the system topology that makes `saw.SaveState()` not behave (I attempted to avoid this issue by saving the state **after** changing the system topology, but that might still make things not behave normally; I'm not too sure.)

mzy2240, 2023-06-03:
Reloading the case is the safest way to iteratively test many scenarios, second by resetting to flat start if no convergence issue.

Weifarers, 2023-06-03:
What would be the preferred method to reload the case? I assumed it would be:

```python
saw.exit()
saw = SAW(FileName=case_path)
saw.SolvePowerFlow()
```

which, to my understanding, effectively creates a brand new SAW object. But replacing `saw.SaveState()` and `saw.LoadState()` with this still results in consistency issues.

mzy2240, 2023-06-03:
right after saw.exit(), can you add a line of saw = []?

Weifarers, 2023-06-03:
Sure. The code is as follows:

```python
def exit_re_open():
    # Opening the case. Note: ESA requires an absolute path, not a relative one.
    case_path = r'C:\Users\trinh7\OneDrive - LLNL\Documents\White Walker\Consequence Enumeration' \
                r'\ESA Test\10k Case\ACTIVSg10k.pwb'

    # Given the case path, we first generate the SAW object.
    saw = SAW(FileName=case_path, pw_order=True)

    # Then we solve the power flow.
    saw.SolvePowerFlow()

    # We then want all the relevant parameters for generators. The SAW command gets us the most important numerical
    # identifiers for generators, but then we augment it with specific fields that we're interested in.
    gen_key_fields = saw.get_key_field_list('gen')
    gen_params = gen_key_fields + ['BusName', 'GenMW', 'GenStatus', 'BusCat']
    gen_data = saw.GetParametersMultipleElement(ObjectType='gen', ParamList=gen_params)

    # Then we consider only generators that are already closed.
    gen_data = gen_data[gen_data['GenStatus'] == 'Closed']

    # Also, we remove the slack bus from consideration.
    gen_data = gen_data[gen_data['BusCat'] != 'Slack']

    # Because generators are mapped using a bus number and generator ID, we zip this information together into a
    # list, and append it to the end of the DataFrame. This makes it easier to reference and outage exactly the
    # generators we want to, especially when there are multiple generators at one bus, so you can't just rely on
    # the bus number. We also use the bus name, just in case as well.
    gen_data['list'] = [list(a) for a in zip(gen_data['BusNum'], gen_data['GenID'], gen_data['BusName'])]

    # We also need the relevant parameters and information for the loads, since we'll be uniformly scaling the load
    # in order to reach convergence.
    load_key_fields = saw.get_key_field_list('load')
    load_params = load_key_fields + ['LoadSMW', 'LoadSMVR']
    load_data = saw.GetParametersMultipleElement(ObjectType='load', ParamList=load_params)

    # Defining the generators we want to outage.
    outage_list = [[' 40843', '1', 'PHOENIX 74 4'], [' 40844', '1', 'PHOENIX 74 5'], [' 10811', '1', 'RICHLAND 4 4'],
                   [' 26134', '1', 'AVILA BEACH 5'], [' 26133', '1', 'AVILA BEACH 4'], [' 11016', '1', 'GRAND COULEE 6'],
                   [' 11018', '1', 'GRAND COULEE 8'], [' 50386', '1', 'DELTA 2 4'], [' 50385', '1', 'DELTA 2 3'],
                   [' 10684', '1', 'MATTAWA 2 5']]

    # We create a copy of the generator information to modify and pass back to PowerWorld.
    gen_copy = gen_data.copy(deep=True)
    # Then we replace all the generator information so that the generators will be open instead of closed.
    gen_copy.loc[(gen_copy['list'].isin(outage_list)), 'GenStatus'] = ['Open'] * len(outage_list)

    # We create a list of load multipliers to test.
    load_multiplier_list = [99, 99.9, 99.99, 99.999]

    print('Load Multiplier List Order: ', load_multiplier_list)
    for load_multiplier in load_multiplier_list:
        # We generate a copy of this data for re-loading purposes.
        load_copy = load_data.copy(deep=True)

        # We then change and confirm the parameter change. Note that command_df requires that the dataframe only
        # contains the key fields and the parameters that are getting changed, so we drop everything else.
        saw.change_and_confirm_params_multiple_element(ObjectType='gen',
                                                       command_df=gen_copy.drop(
                                                           ['GenMW', 'BusName', 'BusCat', 'list'],
                                                           axis=1))

        # Then we try to solve the power flow. If the case converges, we take note of that.
        try:
            load_copy['LoadSMW'] = [float(load_smw) for load_smw in load_copy['LoadSMW']]
            load_copy['LoadSMVR'] = [float(load_smvr) for load_smvr in load_copy['LoadSMVR']]
            load_copy[['LoadSMW', 'LoadSMVR']] *= (load_multiplier / 100)
            saw.change_parameters_multiple_element_df(ObjectType='load', command_df=load_copy)
[... 102 lines cut ...]
        saw.SolvePowerFlow()

    print('----------------------------------------------------------------')

    saw.exit()
```

The same issue persists, where `exit_re_open()` shows:
```
Load Multiplier List Order:  [99, 99.9, 99.99, 99.999]
Successfully solved case with Load Multiplier of 99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.999%
----------------------------------------------------------------
```
while `inv_exit_re_open()` shows:
```
Load Multiplier List Order:  [99.999, 99.99, 99.9, 99]
Unsuccessfully solved case with Load Multiplier of 99.999%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Unsuccessfully solved case with Load Multiplier of 99%
----------------------------------------------------------------
```

mzy2240, 2023-06-03:
How about [99, 99.9, 99, 99.9]?

Weifarers, 2023-06-03:
Similar issue. However, it's occurring to me now looking at the code that by including `continue` during my try-except, in instances where the case does not converge, the system does not get reset as it skips straight to the next iteration. This would explain the behavior, but going to do some debugging, just to make sure.

mzy2240, 2023-06-03:
put change parameters multiple df out the scope of the try except. Not sure if solvepowerflow is the real reason.

mzy2240, 2023-06-03:
Yeah continue should not be used here

Weifarers, 2023-06-03:
Validated. The following two versions:

```python
def exit_re_open():
    # Opening the case. Note: ESA requires an absolute path, not a relative one.
    case_path = r'path\to\ACTIVSg10k.pwb'

    # Given the case path, we first generate the SAW object.
    saw = SAW(FileName=case_path, pw_order=True)

    # Then we solve the power flow.
    saw.SolvePowerFlow()

    # We then want all the relevant parameters for generators. The SAW command gets us the most important numerical
    # identifiers for generators, but then we augment it with specific fields that we're interested in.
    gen_key_fields = saw.get_key_field_list('gen')
    gen_params = gen_key_fields + ['BusName', 'GenMW', 'GenStatus', 'BusCat']
    gen_data = saw.GetParametersMultipleElement(ObjectType='gen', ParamList=gen_params)

    # Then we consider only generators that are already closed.
    gen_data = gen_data[gen_data['GenStatus'] == 'Closed']

    # Also, we remove the slack bus from consideration.
    gen_data = gen_data[gen_data['BusCat'] != 'Slack']

    # Because generators are mapped using a bus number and generator ID, we zip this information together into a
    # list, and append it to the end of the DataFrame. This makes it easier to reference and outage exactly the
    # generators we want to, especially when there are multiple generators at one bus, so you can't just rely on
    # the bus number. We also use the bus name, just in case as well.
    gen_data['list'] = [list(a) for a in zip(gen_data['BusNum'], gen_data['GenID'], gen_data['BusName'])]

    # We also need the relevant parameters and information for the loads, since we'll be uniformly scaling the load
    # in order to reach convergence.
    load_key_fields = saw.get_key_field_list('load')
    load_params = load_key_fields + ['LoadSMW', 'LoadSMVR']
    load_data = saw.GetParametersMultipleElement(ObjectType='load', ParamList=load_params)

    # Defining the generators we want to outage.
    outage_list = [[' 40843', '1', 'PHOENIX 74 4'], [' 40844', '1', 'PHOENIX 74 5'], [' 10811', '1', 'RICHLAND 4 4'],
                   [' 26134', '1', 'AVILA BEACH 5'], [' 26133', '1', 'AVILA BEACH 4'], [' 11016', '1', 'GRAND COULEE 6'],
                   [' 11018', '1', 'GRAND COULEE 8'], [' 50386', '1', 'DELTA 2 4'], [' 50385', '1', 'DELTA 2 3'],
                   [' 10684', '1', 'MATTAWA 2 5']]

    # We create a copy of the generator information to modify and pass back to PowerWorld.
    gen_copy = gen_data.copy(deep=True)
    # Then we replace all the generator information so that the generators will be open instead of closed.
    gen_copy.loc[(gen_copy['list'].isin(outage_list)), 'GenStatus'] = ['Open'] * len(outage_list)

    # We create a list of load multipliers to test.
    load_multiplier_list = [99, 99.9, 99.99, 99.999]

    print('Load Multiplier List Order: ', load_multiplier_list)
    for load_multiplier in load_multiplier_list:
        # At the start of every loop, we exit the file, then re-open it and re-solve the initial power flow.
        saw.exit()
        saw = []
        saw = SAW(FileName=case_path)
        saw.SolvePowerFlow()
        # We generate a copy of this data for re-loading purposes.
        load_copy = load_data.copy(deep=True)

        # We then change and confirm the parameter change. Note that command_df requires that the dataframe only
        # contains the key fields and the parameters that are getting changed, so we drop everything else.
        saw.change_and_confirm_params_multiple_element(ObjectType='gen',
                                                       command_df=gen_copy.drop(
                                                           ['GenMW', 'BusName', 'BusCat', 'list'],
                                                           axis=1))

        load_copy['LoadSMW'] = [float(load_smw) for load_smw in load_copy['LoadSMW']]
        load_copy['LoadSMVR'] = [float(load_smvr) for load_smvr in load_copy['LoadSMVR']]
[... 96 lines cut ...]
    print('----------------------------------------------------------------')

    saw.exit()
```

returns:
```
Inverse Load Multiplier List Order:  [99.999, 99.99, 99.9, 99]
Load Multiplier List Order:  [99.999, 99.99, 99.9, 99]
Unsuccessfully solved case with Load Multiplier of 99.999%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Successfully solved case with Load Multiplier of 99%
----------------------------------------------------------------
Load Multiplier List Order:  [99, 99.9, 99.99, 99.999]
Successfully solved case with Load Multiplier of 99%
Unsuccessfully solved case with Load Multiplier of 99.9%
Unsuccessfully solved case with Load Multiplier of 99.99%
Unsuccessfully solved case with Load Multiplier of 99.999%
----------------------------------------------------------------

Process finished with exit code 0
```

Indicating this was an ordering error more than anything; apologies about that. Just to confirm, the best methodology for doing this sort of analysis is to always exit and re-open the case, correct? There's some performance concerns from me (just in terms of having to re-open a case so many times), but I suppose that's a product of the type of analysis I'm doing, and there isn't a method in which to speed that up?

mzy2240, 2023-06-03:
You can also manually save all the states and status, and restore every time. Most of the time branch, bus, load, and gen.

Weifarers, 2023-06-03:
By that do you mean individually saving those states in dataframes, and then passing them back to the case with `saw.change_and_confirm_params_multiple_element()` or `saw.change_parameters_multiple_element_df()`?

Do you think that would potentially pose problems if I try to re-solve the case from a state that didn't converge? Or are you referring to a different method of doing that?

mzy2240, 2023-06-03:
How about saving the base case?

Weifarers, 2023-06-03:
So something like:

```python
# Opening the case. Note: ESA requires an absolute path, not a relative one.
case_path = r'path\to\ACTIVSg10k.pwb'

# Given the case path, we first generate the SAW object.
saw = SAW(FileName=case_path, pw_order=True)

# Then we solve the power flow.
saw.SolvePowerFlow()
saw.SaveState()

# Code here getting the generator and load data.

# We create a list of load multipliers to test.
load_multiplier_list = [99, 99.9, 99.99, 99.999]

for load_multiplier in load_multiplier_list:
        # At the start of every loop, we reload the base case.
        saw.LoadState()
    
        # Code here to open the generators and uniformly reduce the load.

        try:
            saw.SolvePowerFlow()
            print("Successfully solved case with Load Multiplier of {}%"
                  .format(load_multiplier ))
        except esa.saw.PowerWorldError:
            print("Unsuccessfully solved case with Load Multiplier of {}%".format(load_multiplier ))

```

Will there be any issues with loading the base case after opening generators? The documentation mentions that `saw.LoadState()` does not behave properly if the system topology is changed.

mzy2240, 2023-06-03:
When you save the state, I mean manually use the GetParametersMultipleElement function, you could also save the Status like branch status or gen status.

Weifarers, 2023-06-03:
So you're saying to call `ChangeParametersMultipleElement` for `branch`, `bus`, `load`, and `gen`, and then at the end of every loop call `change_and_confirm_params_multiple_element` or `change_parameters_multiple_element_df`?

mzy2240, 2023-06-03:
Yes

Weifarers, 2023-06-03:
Noted; I'll try that! Would there be a need to re-solve the power flow after calling `change_and_confirm_params_multiple_element ` for those 4 elements? I assume the answer is no, but it wouldn't hurt to do so.

mzy2240, 2023-06-03:
No need if you save enough states

mzy2240, 2023-06-03:
Feel free to re-open if you still encounter issues

closed 2023-06-03 by mzy2240