mzy2240/ESA #110: [SURVEY] What organization are using ESA?

mzy2240, 2023-06-17:
We are interested in learning more about the organizations utilizing ESA and how they are leveraging the package. 

Please leave a comment to let us know the name of your organization. You are also more than welcome to provide a brief description of how you are using ESA. Your feedback is really valuable to us and may very likely impact the future development.

We hope that everyone enjoys using ESA, and your comments will play a significant role in helping us make it even better!

Thanks!

faoh, 2023-06-18:
Princeton University.
I am using ESA to streamline power system analysis and extract graphic properties of power networks.

Weifarers, 2023-06-26:
Lawrence Livermore National Laboratory.

I'm using ESA to automate a large amount of contingencies, and hunting for ones that do not converge. I then go through uniform and geographic-based load scaling in an attempt to make the system converge.

TBendong, 2023-06-26:
The University of Connecticut.
I am using ESA to do batch dynamic simulations. Automation of ESA does help me a lot in this task.