mzy2240/ESA #111: Unable to add a load to a bus that does not already have a load

t-rob124, 2023-06-26:
Hi, I am new to ESA so it is possible that I am missing something in order to add a new load. I am using the example of adding a line to a system and modifying it to be able to add a load. I am able to modify a load that exists in the system, however on a bus where there was not previously a load, when I try to add one it gives me a "No Object Found" error. 
To replicate error, bus 8 of the IEEE 14 bus system is the bus with no load that gives me an error. 
This is my code: 

`#Define the case path to the system
CASE_PATH = r"path\IEEE 14 bus_pws_version_23.pwb"
LOADS_ADDED = r"path\loads_added.csv"

#import the SimAuto wrapper class and initialize an instance
from esa import SAW
import pandas as pd

#Read in new loads file, and define saw 
#the key here for transmission lines was CreateIfNotFound = True
new_loads = pd.read_csv(LOADS_ADDED)
saw = SAW(FileName=CASE_PATH, CreateIfNotFound = True, early_bind = True)

#get key fields from loads (These lines are not strictly necessary in the code, they just were in the load modification example)  
kf = saw.get_key_field_list('load')
load_frame = saw.GetParametersMultipleElement('load', kf + ['LoadSMW', 'LoadSMVR'])

saw.RunScriptCommand("EnterMode(EDIT);")
saw.change_and_confirm_params_multiple_element('load', new_loads)
****This is where it gives the Object Not Found error for bus 8 of the IEEE system, but not bus 4

saw.SolvePowerFlow()
saw.get_power_flow_results('load') `

[IEEE 14 bus test case.zip](https://github.com/mzy2240/ESA/files/11873698/IEEE.14.bus.test.case.zip)

mzy2240, 2023-06-26:
Hi @t-rob124, I think you are missing one Status field. When adding a new object, make sure you have specified not only key fields but also required fields. This is a hard requirement of PowerWorld simulator, which means it would also apply to the GUI, the aux file, and of course ESA.

Weifarers, 2023-06-26:
Curious if there was a way to identify the necessary fields from ESA; I know the key fields are fields most necessary for finding any particular element in the system, but in the [example of adding lines](https://mzy2240.github.io/ESA/html/snippets.html#add-lines-to-case), it's not explicitly mentioned (as far as I can tell) how to find the required fields for adding new objects.

[image]

It seems like in PowerWorld they identify them by highlighting the column names in green in Explorer. Is there an associated flag in ESA for those types of fields?

mzy2240, 2023-06-27:
@Weifarers You can actually find them in the model explorer. Assuming those fields would not change too frequently, we probably could hardcode those fields for common objects in ESA to give more hints of the error.
[image]

Weifarers, 2023-06-27:
@mzy2240 This is very useful information, thanks! :)

mzy2240, 2023-06-27:
@t-rob124 Feel free to re-open the issue if you still have questions.

closed 2023-06-27 by mzy2240