mzy2240/ESA #112: How to open RawFile Version 26?

SaeedNsh, 2023-07-10:
Hi everyone,
Anybody knows how Can I open a powercase in rawfile format with version26? 
The following code will open Version33 but when I tried for Version 26 it comes with an error.

from esa import SAW
saw=SAW(CASE_PATH)

SaeedNsh, 2023-07-10:
> Check this out:

Thank you for prompt answer.
I have tried it but it doesn't work even with 'PTI26'.
Would you please take a look at the line of the code?

saw=SAW.OpenCaseType('CASE_PATH','PTI26','')

And this is the error I receive.
Note that I just want to load a powercase rawfile which has version26.

Regards,
Saeed

SaeedNsh, 2023-07-10:
I received this error:
'str' object has no attribute 'pwb_file_path'

mzy2240, 2023-07-10:
```python
saw=SAW(fake_case) # load any case first, PTI v33 or PWB file
saw.OpenCaseType(real_case, "PTI26") # now load the case that is PTI26
```
Currently there is no good way to load a case with specified file type, which could be easily fixed in the __init__ method by introducing a new optional argument `file_type`.

SaeedNsh, 2023-07-10:
from esa import SAW
saw=SAW(CASE_PATH)  # load a case first with PTI v33
saw.OpenCaseType(myfilename,"PTI26") # Then load the real file with PTI v26

The second line get loaded but still the following error comes up for third line:
esa.saw.PowerWorldError: OpenCaseTypeSimAuto: Invalid Creation of Loads Transactions parameter.

SaeedNsh, 2023-07-10:
The alternative way to load a version26 Rawfile is the following line:
simauto_obj.RunScriptCommand("OpenCase(" + myfilename + ", PTI26)")

It is working, However, I need to use ESA package.

mzy2240, 2023-07-11:
Then try this
```python
from esa import SAW
saw = SAW(CASE_PATH) # load a case first with PTI v33
saw.RunScriptCommand("OpenCase(" + myfilename + ", PTI26)")
```

mzy2240, 2023-07-11:
> The second line get loaded but still the following error comes up for third line:
> esa.saw.PowerWorldError: OpenCaseTypeSimAuto: Invalid Creation of Loads Transactions parameter.

Do you mind to share the PTI26 case? `OpenCaseType` works fine in our test.

SaeedNsh, 2023-07-11:
Sure.
Please find it attached.
It would be great if you check it with your code.

SaeedNsh, 2023-07-11:
[Original200_1_0.856147436030558.zip](https://github.com/mzy2240/ESA/files/12009914/Original200_1_0.856147436030558.zip)
It is in the zip file attached.

mzy2240, 2023-07-11:
`saw.OpenCaseType(r"C:\Users\\Downloads\Original200_1_0.856147436030558\Original200_1_0.856147436030558.raw", FileType="PTI26", Options="NO")` you need to set `LoadTransaction` to `NO`

SaeedNsh, 2023-07-11:
Zeyu,
Thanks for your support. It is working now.

mzy2240, 2023-07-11:
Feel free to leave a comment if you found ESA useful: https://github.com/mzy2240/ESA/issues/110

closed 2023-07-11 by mzy2240